robinhedwards/UnoCart-2600 #4: Will the PCB files be made public?

Arduingo, 2018-05-10:
Great work, and thanks to make it public, are you going to release de PCB files? in case no, do you mind if someone else release his PCB design and link to this project?

robinhedwards, 2018-05-11:
The PCB design will be made public at some point in the future - I've been redesigning it to fit in a cart shell and use an SD card rather than a microSD.

These will initially be produced by MacRorie in the USA for sale, and possibly someone in europe too.
I'll update the page when this happens.

root42, 2019-05-20:
Would love to see this happen, or have the ability to buy it from someone in Europe. Any chance of progress on this?

ranger81, 2020-04-03:
@robinhedwards Is there any update on releasing the PCB files or schematics?
Thanks

TabajaraLabs, 2020-04-03:
Looking for the PCB files too =)

tdididit, 2020-04-03:
@robinhedwards If you agree I will make public my PCB design with USB ISP circuit addon

ranger81, 2020-04-03:
@tdididit Would be greatly appreciated. Schematic would be fine for me - also I would be very interested about the USB ISP circuit addon. Please let me know, if your schematics are available somewhere.

tdididit, 2020-04-03:
@ranger81 I didn't seen a schematics, but only thing what you need to make it is reading with understanding UnoCart2600 manual file. Like I wrote before I can make my github repository public if @robinhedwards will agree

mase76, 2022-09-01:
Can you publish the schematics?

manotroll, 2022-09-15:
up

manotroll, 2024-01-13:
?