robinhedwards/UnoCart-2600 #12: Issues with my build

Spyke1975, 2020-06-14:
Thank you for sharing this nice work!
I build a cart, but it didn´t work correct. I build this version:
https://circuit-board.de/forum/index.php/Thread/26071-UnoCart-2600-ein-Selbstbau-SD-Karten-Atari-2600-bzw-7800-Modul/
with the black Chinese STM32F4XX.
The cart boots and I can see the menue for a short time, it also reads my SD, because I can see the content. But after a few seconds it ens up in blue screen.
Any ideas what the problem is?

Thanks in advance and best regards
Marc

robinhedwards, 2020-06-15:
Hi Marc,

Does it always blue screen at the same point? Maybe check your jumper wires
- I've had issues like this before with bad connections/partially broken
wires.

Robin

Spyke1975, 2020-06-15:
Hi Robin,

rhanks for reply.
yes it's allways at the same point.
2 or 3 seconds after the menu appears.
I checked all conections and they seem to be ok. Do you think the Problem is in the connections for the board or the SD-modul?

In the short time I have tried to start a game and it works, but end up in blue scrreen. Maybe the board is faulty. 
Marc