robinhedwards/UnoCart-2600 #13: Display only "SD" on blue screen

ranger81, 2020-06-18:
I have built an UnoCart and flashed the latest binary. Without plugged in SD-Cart, the menu shows up and displays "CANT READ SD", so far so good.
When I insert the SD Cart (FAT32), the screen is filled in blue and displays only "SD" in big letters.

[image]

Do you have any advice, what could be wrong?

Thanks!

robinhedwards, 2020-06-19:
As a first step, it's probably worth trying a different SD card with only a
few files on it.

ranger81, 2020-06-28:
Thanks for you support, I ordered some STM32F407 but got STM32F207 delivered. Didn't check them before soldering. Now I got proper STM32F407 and it works as expected. Sorry to bother you!

closed 2020-06-28 by ranger81

robinhedwards, 2020-06-28:
Okay, I would never have guessed you'd done that! Glad you got it working.