robinhedwards/UnoCart-2600 #20: Is there an STL file available?

jgrillout, 2022-01-07:
I'm new to this whole UniCart product. It is very cool. I want to 3d print my own cartridge. Is there a STL file available?  Also where can I buy the rest of the hardware for this project. I saw a video, I can't remember if it was for Uni or harmony, that mentioned the project became open source. Does this apply to code only or hardware too?  

Thank you,
John

ojaksch, 2022-01-07:
Can't remember where my case came from but found this: https://www.thingiverse.com/thing:4580701

jgrillout, 2022-01-07:
thanks Larry !

and-pad, 2022-04-05:
> I'm new to this whole UniCart product. It is very cool. I want to 3d print my own cartridge. Is there a STL file available? Also where can I buy the rest of the hardware for this project. I saw a [...]

Hi John, I have finished some parts and it works, if you are still interested, I don't know if you have got the drawing of the atari cartridge.
If you are interested I leave you my email: andrestor@gmail.com
Andres Padilla