robinhedwards/UnoCart-2600 #21: Hardware alternative?

Lalarian, 2022-04-02:
Hello!

As the Stm32f407 dev board has become expensive (35€ including shipping for the cheapest offer I could find), I wonder if there's an more affordable alternative hardware which the firmware could easily be translated, to?

darleiv, 2022-06-18:
@DirtyHairy

darleiv, 2022-06-18:
@DirtyHairy

darleiv, 2022-06-18:
@robinhedwards

MrGasS, 2022-11-08:
I was thinking the same thing, an STM32 BluePill port would be beautiful as the Discovery board is crazy expensive.

gtortone, 2023-07-08:
Hi,
in next weeks I will start to work (in my spare time) on a porting for ESP32 based on ESP32-S2-mini that is widely
available on online stores at very low price. I have in plan to design a board with 5V - 3.3V level converter and a "socket" to plug-in ESP32-S2-mini board...

MrGasS, 2023-07-09:
> Hi, in next weeks I will start to work (in my spare time) on a porting for ESP32 based on ESP32-S2-mini that is widely available on online stores at very low price. I have in plan to design a board with 5V - 3.3V level converter and a "socket" to plug-in ESP32-S2-mini board...

thank you a lot, can't wait to see your ESP32 port :)
(Forza Napoli)

manotroll, 2024-01-13:
Raspberry Pi Pico does not directly support up to 30V
and wouldn't it still be cheaper than an esp32?

gtortone, 2024-04-19:
Hi, 
if you are interested to this development, I completed Raspberry Pico porting of PlusCart:

https://github.com/gtortone/United-Carts-of-Atari

manotroll, 2024-04-19:
> Hi, if you are interested to this development, I completed Raspberry Pico porting of PlusCart:
> 
> https://github.com/gtortone/United-Carts-of-Atari

Do you have a list of the components that are used?
and there is no way to use the SD or wifi?

gtortone, 2024-04-20:
@manotroll the PlusCart with RP2040 provides microSD and wifi interface. For wifi it uses the default PlusCart store called PlusStore where you will find hundreds of classic games and homebrew for Atari 2600...

to build a PlusCart board with RP2040 you need 

- a Raspberry Pico board, not the "green" because a "purple" one exists (look on AliExpress) that has more pins available on header
- a microSD breakout board (Amazon or AliExpress)
- an ESP-01 module (Amazon or AliExpress)

In a few days I will write some documentation on the GitHub repository (pinout, build instructions, ...), but in the meanwhile let me know if you need help !